klatremis/esphome-for-deye #2: Number to time

kababoom, 2023-02-06:
Thank you for creating this repo. Nice clean way of communicating with the inverter, I like it...

Tried with a 8266 first, lights of the tll-rs485 were flashing but nothing received, also not with logging disabled (as per esphome wiki). 

ESP32 WROOM 4Mb Devkit V1 working first try..

There's one thing that I don't like, it's the time value, but no matter what lambda I try or HA sensor I create I can't get it to any decent time format.

Currently using:

```
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    id: sun3p_Time_point_1
    name: "sun3p-Time point 1 start"
    min_value: 0
    max_value: 2400
    step: 100
    address: 148
    value_type: U_WORD
```

MyHekla, 2023-02-14:
Did you change the RX/TX pin numbers when you tried the 8266?

kababoom, 2023-02-19:
Yes, tried 1/3 and 13/15..

MyHekla, 2023-02-19:
I've found that the 8266 is pretty picky on pin naming. For RX/TX I would use GPIO3/GPIO1 respectively.

kababoom, 2023-02-20:
The ESP32 is working fine so I leave it at that.

About the issue, I've created a few select entries with just the hours since this is the only thing I'm interested in..

```
select:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "sun3p-Limit control mode"
    id: sun3p_limit_control_mode
    address: 142
    value_type: U_WORD
    optionsmap:
      "Allow Export": 0
      # "Essentials": 1
      "Zero Export": 2

  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    id: sun3p_Time_point_1_start
    name: "sun3p-Time point 1 start"
    address: 148
    value_type: U_WORD
    optionsmap:
      "00:00": 0
      "01:00": 100
      "02:00": 200
      "03:00": 300
      "04:00": 400
      "05:00": 500
      "06:00": 600
      "07:00": 700
      "08:00": 800
      "09:00": 900
      "10:00": 1000
      "11:00": 1100
      "12:00": 1200
      "13:00": 1300
      "14:00": 1400
      "15:00": 1500
      "16:00": 1600
      "17:00": 1700
      "18:00": 1800
      "19:00": 1900
      "20:00": 2000
      "21:00": 2100
      "22:00": 2200
      "23:00": 2300
      "24:00": 2400
```

closed 2023-02-20 by kababoom