klatremis/esphome-for-deye #3: Random measures from SUN-12K

AndihX, 2023-02-23:
Hi,

We've just flashed a ESPHome on a ESP32, and used your configuration. It's initiated as it should, however, it's reporting very random-like numbers on each sensor.

[image]
[image]

We've followed the wirering diagram, and it's clear that it's OK.

After the ESP32 has been rebooted, we're back to normal numbers for a period of time.

Is this a timing or signal issue? Have you experinced simular problems? (We just tested with 20s interval)

Is is triggered by a specific event on the DEYE, or?

Here's a correct "load" with numbers:
[image]

AndihX, 2023-02-23:
Replugging the power to the RS485>TTL module board, things works fine again.

This time it worked for appx. 10 minutes, and ended up like this:
```log
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 4 start': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 5 start': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 6 start': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 8000.00
[20:19:13][D][number:012]: 'sun12k-Time point 1 power': Sending state 8000.000000
[20:19:13][D][modbus.number:023]: Number new state : 8000.00
[20:19:13][D][number:012]: 'sun12k-Time point 2 power': Sending state 8000.000000
[20:19:13][D][modbus.number:023]: Number new state : 8000.00
[20:19:13][D][number:012]: 'sun12k-Time point 3 power': Sending state 8000.000000
[20:19:13][D][modbus.number:023]: Number new state : 8000.00
[20:19:13][D][number:012]: 'sun12k-Time point 4 power': Sending state 8000.000000
[20:19:13][D][modbus.number:023]: Number new state : 8000.00
[20:19:13][D][number:012]: 'sun12k-Time point 5 power': Sending state 8000.000000
[20:19:13][D][modbus.number:023]: Number new state : 8000.00
[20:19:13][D][number:012]: 'sun12k-Time point 6 power': Sending state 8000.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 1 capacity': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 500.00
[20:19:13][D][number:012]: 'sun12k-Time point 2 capacity': Sending state 500.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 3 capacity': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 4 capacity': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 5 capacity': Sending state 0.000000
[20:19:13][D][modbus.number:023]: Number new state : 0.00
[20:19:13][D][number:012]: 'sun12k-Time point 6 capacity': Sending state 0.000000
[20:19:13][D][switch:056]: 'sun12k-Time point 1 charge enable': Sending state OFF
[20:19:13][D][switch:056]: 'sun12k-Time point 2 charge enable': Sending state OFF
[20:19:13][D][text_sensor:067]: 'sun12k-Running Status': Sending state 'standby'
[20:19:13][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[20:19:13][D][sensor:127]: 'sun12k-active power generation of today': Sending state 0.00000 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 2467.00
[20:19:14][D][sensor:127]: 'sun12k-Total charge of the battery': Sending state 246.70000 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 2391.00
[20:19:14][D][sensor:127]: 'sun12k-Total discharge of the battery': Sending state 239.10001 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 1.00
[20:19:14][D][sensor:127]: 'sun12k-Daily Energy Bought': Sending state 0.10000 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 113.00
[20:19:14][D][sensor:127]: 'sun12k-Daily Energy Sold': Sending state 11.30000 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 5204.00
[20:19:14][D][sensor:127]: 'sun12k-Total Energy Bought': Sending state 520.40002 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 2467.00
[20:19:14][D][sensor:127]: 'sun12k-Total Energy Sold': Sending state 246.70000 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 2018906880.00
[20:19:14][D][sensor:127]: 'sun12k-Total Consumption': Sending state 201890688.00000 kWh with 2 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 44474.00
[20:19:14][D][sensor:127]: 'sun12k-Daily Production': Sending state 4447.39990 kWh with 1 decimals of accuracy
[20:19:14][D][modbus_controller.sensor:025]: Sensor new state: 6832.00
[20:19:14][D][sensor:127]: 'sun12k-Total PV Production': Sending state 683.20001 kWh with 1 decimals of accuracy
[20:19:15][D][modbus_controller.sensor:025]: Sensor new state: 2760.00
[20:19:15][D][sensor:127]: 'sun12k-Heat sink temperature': Sending state 176.00000 °C with 2 decimals of accuracy
[20:19:15][D][modbus_controller.sensor:025]: Sensor new state: 1402.00
[20:19:15][D][sensor:127]: 'sun12k-Failure status of communication board': Sending state 1402.00000  with 0 decimals of accuracy
[20:19:15][D][binary_sensor:036]: 'sun12k-Turn off/on status': Sending state OFF
[20:19:15][D][binary_sensor:036]: 'sun12k-AC INV relay': Sending state OFF
[20:19:15][D][binary_sensor:036]: 'sun12k-AC Load relay Reserved': Sending state ON
```

Note the "standby" signal. I'm not sure if that just a faulty read, or if it's actually a proper state returned by the MODBUS-interface from the DEYE?

pvprodk, 2023-03-28:
> Note the "standby" signal. I'm not sure if that just a faulty read, or if it's actually a proper state returned by the MODBUS-interface from the DEYE?

The standby is a valid Running Status, when you dont have a battery installed, and the sun is down. Mine does the same:
[image]

keldbroe, 2023-06-22:
I've got the same problem.
Have you solved it, or do you still have the issue?

AndihX, 2023-06-23:
No solution yet, though i am glad that i am not the only one. What i have
figured out, is that it is the 12 to 5v board that is some how not stable.
My temporarily solution was to use a 12 to 5V usb charger for a Mobile
phone. It does not look nice, but it helps for now. I have tested 3 12 to
5v boards so far, without sucess.

keldbroe, 2023-06-26:
Strange thing. I've tried to replace the PSU to a normal 230V-USB adaptor, and it seems to fix the problem. 
I've noticed that, when I have the problems the TTL/RS-485 board is VERY hot. So hot so I can burn my finger. 
When it worked the temperature was not noticeable.
Could you try that?
Or as a minimum try to touch the TTL/RS-485 board briefly.

sker65, 2023-08-25:
I also having problems reading the modbus. Basically it works, but the log shows CRC errors and there always strange outliers in the sensor values e.g. heatsink temp: 40 -> 200 -> 40.

Tried different cable, different RS485 board, different version of arduino framework. basically nothing changed. You cannot reliably read the data over more than a couple of minutes

fabhund, 2023-08-26:
> Strange thing. I've tried to replace the PSU to a normal 230V-USB adaptor, and it seems to fix the problem. I've noticed that, when I have the problems the TTL/RS-485 board is VERY hot. So hot so I [...]

I have mixed results: Im using a ESP32-Wroom in a D1-mini board, works fine on 3 installations. Now I replaced 1 in the Deye inverter to a ESP32U-Wroom (U for external antenna) this made the TTL board run hot instantly, running on the same internal power and a buck converter to 12->5V. Now I tried an external 230V-5V USB charger, and temperatue is down again.

klatremis, 2023-11-14:
I have reports on that there is different models of the TTL/RS-485 boards. Even that they look like the same, they act different.
Some people have reportet that the boards from one supplier died after a few days. I think its all about the component quality.

My suggestion is:
- Use external usb supply, or a step down adaptor from 12V in the inverter to 5V Usb
- Try another supplier of rs485/ttl board

closed 2023-11-14 by klatremis