klatremis/esphome-for-deye #5: Question to modbus operation ?

fabhund, 2023-03-09:
Dear Team.
I see the update frequency is 15s, which is great for live-feed.

However, does it also write back the values every 15s ?
My concern is, if the values are also constantly written over and over (every 15s) and eventually wear out the memory flash-chip in the Deye inverter.

But I'm unsure how the detailed workings are with the modbus and state changes in ESPHome / Homeassistant.
The optimum would be to read at every 15s and write only updated values back to inverter.
Thank you.

PS: My first github post ever, so please be gentle :-)

kababoom, 2023-03-16:
I did wonder the same thing but it looks like the esp only writes when there's something changed.

You can check it yourself by looking at the update time of one of the write registers. 

For example, mine says an hour ago at this moment.

[image]

pvprodk, 2023-03-28:
> I did wonder the same thing but it looks like the esp only writes when there's something changed.
> 
> You can check it yourself by looking at the update time of one of the write registers.
> 
> For example, mine says an hour ago at this moment.
> 
> [image]

This screenshot does not prove that values are not written to the flash-chip with same value every 15 minutes. It just shows that from a Home Assistant recorder perspective, it has been one hour since the value changed.

kababoom, 2023-04-11:
> > I did wonder the same thing but it looks like the esp only writes when there's something changed.
> > You can check it yourself by looking at the update time of one of the write registers.
> > For example, mine says an hour ago at this moment.
> > [image]
> 
> This screenshot does not prove that values are not written to the flash-chip with same value every 15 seconds. It just shows that from a Home Assistant recorder perspective, it has been one hour since the value changed.

Correct and was aware of that, just one thing to add. If you change a writable register it appears in the logging, like it does for the reading ones every 15 seconds, then there's a 'remove from queue' message and it will not show up anymore. 
This also indicates it only writes when there's something to write, but still not 100%, perhaps there's info on it at esphome.

closed 2023-06-29 by klatremis