klatremis/esphome-for-deye #7: Home Assistant Energy Dashboard Error

linuxdep, 2023-03-13:
Hi, nice project, many thanks.

I wanted to insert the total PV production in the Energy Dashboard, but this fails because there seems to be no parameter "last_reset" for it. How can I adjust this so that the data arrives in the dashboard?

[image]

gmidohan, 2023-03-13:
Maybe this is helpful: different device same problem 
[https://community.home-assistant.io/t/energy-dashboard-last-reset-missing/406733](https://community.home-assistant.io/t/energy-dashboard-last-reset-missing/406733)

gplex83, 2023-03-13:
change state_class: "measurement"
to   state_class: "total_increasing"

  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "sun12k-Total PV Production"  #Cumulative Production 
    id: sun12k_Total_PV_Production
    register_type: holding
    address: 534
    unit_of_measurement: "kWh"
    state_class: "total_increasing"
    device_class: energy
    accuracy_decimals: 1
    value_type: U_DWORD_R
    skip_updates: ${skip_updates}
    filters:
      - multiply: 0.1

closed 2023-03-13 by linuxdep