klatremis/esphome-for-deye #8: Modbus-register

dowermyr, 2023-03-22:
Hi! Do you have the modbus-register for this? I wantto add som points

dowermyr, 2023-04-08:
Trying to find something that would trigger discharge of battery to sell to grid. Any idea?

kababoom, 2023-04-11:
I use this for the ESP:
```

select:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "sun3p-Limit control mode"
    id: sun3p_limit_control_mode
    address: 142
    value_type: U_WORD
    optionsmap:
      "Allow Export": 0
      # "Essentials": 1
      "Zero Export": 2

```
Then something like this for the card:
```

type: entities
entities:
  - entity: select.sun3p_limit_control_mode
    name: Limit Mode
  - entity: number.sun3p_maximum_battery_discharge_current
  - entity: number.sun3p_maximum_battery_charge_current

```

[image]

dowermyr, 2023-05-28:
Hi! Trying to find adress for peak shaving. Both on/off and set the value.
Do you know if it exist?

tomatensaus, 2023-05-31:
I have a single phase deye inverter and peak shaving

enable is at 
address: 280
optionsmap:
      "None": 0
      "Enable": 0x0100

Just dump all registers.... change setting and dump them again... slow process but you can reverse engineer it

closed 2023-06-29 by klatremis