klatremis/esphome-for-deye #11: Duplicate modbus command found

manalishi70, 2023-04-21:
I got this message:
<html>
<body>

11:27:41 | [W] | [modbus_controller:113] | Duplicate modbus command found: type=0x3 address=676 count=4

I get this message for many registers. Do you have a clue what causes this ?

</body>
</html>

manalishi70, 2023-04-22:
It's been a HW issue. I switched RS485 Adapters
from  [https://www.amazon.de/-/en/DollaTek-MAX485-Module-RS-485-Development/dp/B07DK4QG6H/ref=pd_bxgy_img_sccl_2/261-1531827-5478614?pd_rd_w=xM2UP&content-id=amzn1.sym.1fd66f59-86e9-493d-ae93-3b66d16d3ee0&pf_rd_p=1fd66f59-86e9-493d-ae93-3b66d16d3ee0&pf_rd_r=55RE4DJDEPXP1NY7QFVE&pd_rd_wg=kzFXG&pd_rd_r=9dee2802-eb23-416c-a04f-4d146e343b61&pd_rd_i=B07DK4QG6H&psc=1](url)

 to (https://www.amazon.de/-/en/dp/B09VGJCJKQ?psc=1&ref=ppx_yo2ov_dt_b_product_details )

I have to use 5V. 3.3V didn't give any sensor results

keldbroe, 2023-06-26:
Strange thing. I've tried to replace the PSU to a normal 230V-USB adaptor, and it seems to fix the problem.
I've noticed that, when I have the problems the TTL/RS-485 board is VERY hot. So hot so I can burn my finger.
When it worked the temperature was not noticeable.
Could you try that?
Or as a minimum try to touch the TTL/RS-485 board briefly.

bagges, 2023-07-24:
This also happend to me, but it was mostly related to bad power supply. CN2 7/8 pins are also just rated about 100mA.

What also sometimes helpes is to add 120 ohm termination on the rs485 module. Most modules have a solder bridge for the termination.

MyHekla, 2023-07-30:
> This also happend to me, but it was mostly related to bad power supply. CN2 7/8 pins are also just rated about 100mA.
>

well that explains the voltage drops and noise I have been measuring. Powering via CN2 seems to have been a bad idea.

klatremis, 2023-11-14:
As said in https://github.com/klatremis/esphome-for-deye/issues/3

My suggestion is:

Use external usb supply, or a step down adaptor from 12V in the inverter to 5V Usb
Try another supplier of rs485/ttl board

closed 2023-11-14 by klatremis