klatremis/esphome-for-deye #12: after powercycle incorrect values are sent

linuxdep, 2023-04-22:
Since I still some Umbaue and test, I switch off the Deye or also times the ESPhome device, after switching on, however, irrwitzig high values are sent. Could not this be turned off?

klatremis, 2023-06-29:
I have tryed the same thing. I dont know what to do about it.

pvprodk, 2023-06-29:
I have tried similiar behaviour on Growatt inverters, where the ESP device powercycles together with the inverter. In that case the following "delay" solves the problem, but that would mean that on Deye you had to power the ESP in a way that it also powercycles with the inverter.

`esphome:
  name: ${device_name}
  on_boot:
    priority: 600
    then:
      - lambda: |-        
          delay(60000);
`

olesj33, 2023-11-09:
[image]

I have the same problem with wrong data from 00:00 to 01:20 each day .

klatremis, 2023-11-14:
> [image]
> 
> I have the same problem with wrong data from 00:00 to 01:20 each day .

This is caused by bad power supply.

Try with external usb supply.

klatremis, 2023-11-14:
> I have tried similiar behaviour on Growatt inverters, where the ESP device powercycles together with the inverter. In that case the following "delay" solves the problem, but that would mean that on Deye you had to power the ESP in a way that it also powercycles with the inverter.
> 
> [image]

Have anyone tryed this? Does it remove wrong data when the esp is starting?

Xabrim, 2024-03-29:
I have the same issue. During nighttime I have 100k Watt PV Production. It totally messes up my statistics. Anyone found a solution?

olesj33, 2024-03-29:
Try a another external power supply,  I had the problem using a cheap one internal.

Xabrim, 2024-03-29:
> Try a another external power supply,  I had the problem using a cheap one internal. 

Has it also occured irregularly? It hasn't happened for a while.

olesj33, 2024-03-29:
> > Try a another external power supply,  I had the problem using a cheap one internal.
> 
> Has it also occured irregularly? It hasn't happened for a while.

Yes there could go days or weeks between.