klatremis/esphome-for-deye #16: Ingen kommunikation med flere invertere

HansPeterHaastrup, 2023-07-24:
Jeg har 2 invertere (Addr. 01 og 02).
Når jeg benytter Mod(e)BUS stikket på masteren, kommer der intet svar. benytter jeg derimod stikket på slave-inverteren, får jeg fint svar - men de data jeg får, er fra slave inverteren! Dette til trods for at jeg beder om data fra adresse 0x1.
Yderligere kan jeg se at hvis jeg ændrer f.eks. time-of-use, så hopper de ændrede værdier tilbage, når masteren synkroniserer data ud.

Jeg har en mistanke om at dette repo kun virker med enkeltstående invertere, men jeg HÅBER at blive modbevist! er der andre i en lignende situation der har haft held med at kommunikere med masteren?

Alphaemef, 2023-07-24:
Hej Hans Peter. 

Deye har en lidt ærgerlig nomenklatur for modbus. Det er ikke samme modbus de kører og kommunikerer på. Du er nødt til at have en ESP32 på hver inverter, og så summere de entitieer du har brug for at summere. Husk hver indstilling du ændrer "fordobler" sig via de to invertere.

HansPeterHaastrup, 2023-07-24:
@Alphaemef mange tak for den info! Jeg går straks i gang med at hækle endnu et stykke hardware!

Men mit problem består - jeg kan ikke få masteren til at svare. Som i overhovedet ikke.
Jeg er mystificeret, for min espHome fungerer netop fint på slaveenheden.

Jeg kan prøve at bytte om på forholdet mellem master og slave, det kunne måske kaste mere lys over sagen

Alphaemef, 2023-11-09:
Hej @HansPeterHaastrup fik du nogensinde løst det med modebus porten på din master ? Defekt controller ?

HansPeterHaastrup, 2023-11-10:
> Hej @HansPeterHaastrup fik du nogensinde løst det med modebus porten på din master ? Defekt controller ?

Hej @Alphaemef  - modebus porten virker af en eller anden grund ikke på min ene inveter.
Løsningen er at lave en MODBUS/CANBUS splitter som kan sættes i BMS stikket. Det virker, fordi alle pins er ført ud til begge stik, men det er virkelig ikke godt dokumenteret.
Heldigvis er den inverter der har problemer, min sekundære, så jeg har ikke noget siddende i BMS stikket, men jeg har lavet en splitter alligevel.
Beskeden fra strømitag, var at det var den officielle løsnng fra Deye.

Jeg har overvejet at lave en indstøbt løsning med et lille print og 3 RJ-45 stik hvis der er andre i samme situation...

HansPeterHaastrup, 2023-11-10:
Jeg lukker dette issue - se første svar fra @Alphaemef - der skal benyttes en ESP32 på hver inverter i systemet.
Hvis der er problemer på MODBUS porten, kan der indsættes en splitter på BMS porten, så både CAN og MODBUS kan kobles ind (de bruger forskellige pins)

closed 2023-11-10 by HansPeterHaastrup

Alphaemef, 2023-11-10:
hmm, det lyder som et defekt bottom print. Er stødt på en inverter hvor jeg tror controlleren er defekt. For kan godt læse på BMS porten, men kan ikke skrive til den.

HansPeterHaastrup, 2023-11-10:
Jeg har haft printet ude og måle det igennem - der var ikke noget der så underligt ud der, men det kan også være jeg ikke har fanget det.

Det er ved at være et godt stykke tid siden jeg havde fat i strømitag, men BMS porten er read-only, så vidt jeg husker. der var i alt fald en af dem der var det, fik jeg at vide.

Alphaemef, 2023-11-10:
Ahh ja, det kan godt forklare det. Så må det være to forskellige controllere.