klatremis/esphome-for-deye #17: RS485/TTL-board brænder ofte af.

zaprodk, 2023-07-29:
Hej. Der er rigtig mange folk der oplever af RS485/TTL board brænder af da der står en spænding hen over modulet (lækstrøm) og så futter chippen af. Kan du ikke skrive i vejledningen at man bør lymuse GND tværs over modulet fra side til side? Det løser problemet.

MyHekla, 2023-07-29:
anvender du separat strømforsyning til ESP og anvender du samme til modulet og hvilket modul anvender du?

zaprodk, 2023-07-29:
> anvender du separat strømforsyning til ESP og anvender du samme til modulet og hvilket modul anvender du?

Jeg har ikke selv setup'et, men formoder at folk følger guiden her på siden. X/Y-kondensatorer i netfiltet på de to enheder kan skabe lækstrøm der brænder chippen af.
Jeg er bare elektroniknørd der hjælper med at finde ud af hvorfor tingene går galt en gang imellem :)

MyHekla, 2023-07-29:
Hvis der anvendes det setup der er beskrevet her er de 2 GND allerede forbundet - gennem inverteren. Den GND der er på rs485 siden er 'kun' en over/under beskyttelses GND - se link
https://electronics.stackexchange.com/questions/244425/how-is-this-rs485-module-working
 så det har intet på sig. Det er nok mere, som det også er beskrevet i en anden rapport' at spændingsforsyningen ikke er god nok. Jeg har selv samme problem og har pt en datalogger på. Jeg påtænker en løsning hvor jeg tager strøm fra min batteripakke.

zaprodk, 2023-07-29:
Fair nok. Jeg tror at folk forsyner ESP'en med en USB-strømforsyning eller noget andet HA-relareret hardware, og det er her balladen opstår. Hvis man tager strømforsyningen fra inverteren selv er der selvfølgelig fælles GND, men i tilfælde hvor folk ikke gør det, der sker sære ting.

MyHekla, 2023-07-29:
Det er stadig ikke det der er problemet/løsningen. GND er ikke nødvendig med et balanceret signal. GND er kun til beskyttelse af  ind/udgang på 485 siden.

Breinholm-coder, 2023-09-04:
@MyHekla 
Det er altså ikke korrekt: Uden GND vil de to A og B signaler flyde uden reference - Det er en misforståelse at RS485 ikke behøver GND. Parsnoning og differenssignaler hæver støj-imuniteten.

zaprodk, 2023-09-05:
@Breinholm-coder exactly ...
Det er det samme som CAN-bus i biler. GND er nædvendigt, ellers kunne der være en høj spænding imellem enhederne der brænder line-drivers af. Og det er helt sikkert det der sker med disse boards. Udstyret i hver ende er tøjret med forskellig reference og pludselig står der 127-ish volt imellem to steder, og så står line-driveren af.

Breinholm-coder, 2023-09-05:
> @breinholm <https://github.com/breinholm>
> Det er fuldstændig korrekt! Det er netop det der er ideen i et ballanceret
> signal. Det KAN leve uden fælles GND. Man kan vælge at benytte fælles GND
> på den ballancerede del, men det ændrer ikke på signal kvaliteten.
>
> @zaprodk <https://github.com/zaprodk>
> over/underspændings beskyttelse løses med clamping dioder - og det er det
> på samtlige af de boards jeg har set fra Kina.
>
> I bør begge læse op på ballacerede (ballanced signals) (samt datasheet for
> MAX485) - det har været anvendt i åtier i tele-sektoren og 'senest' i
> ethernet.
>
> Denne tråd startede iøvrigt mad om man skulle 'ødelægge' den galvaniske
> adskillelse de er på modulet.

Hej MyHekla,

Tror vi er næsten enige - undtagen hvad GND gør for signalkvaliteten: Er
der ingen common-mode spændingsforskel imellem de kommunikerende parter vil
en GND forbindelse ikke ændre noget. Men er der for eksempel over 7 V
common-mode støj imellem parterne vil en GND forbindelse være guld værd til
at nedsætte common-mode forskellen. TIA RS-485 standarden foreskriver
faktisk en fælles reference forbindelse.

MVH / Peter Breinholm

closed 2023-11-14 by klatremis