klatremis/esphome-for-deye #19: Issues with ESP9266

pollierer, 2023-09-09:
Not sure if this is a case to open an issue, if not I´m sorry;
I tried different configurations with a D1 Mini ESP8266. It always ends up in a boot loop.
Eg I also tried to use UART1 for logging and RX/TX Pins GPIO3 and GPIO1 for the modbus connection without success,
Is it possible to use the ESP8266 and I´m doing it wrong or do I need an ESP32 anyway?
Thanks for pointing me into the right direction!

MyHekla, 2023-09-09:
The 8266 can only be used with a very limited number of functions.

pollierer, 2023-09-09:
> The 8266 can only be used with a very limited number of functions.

Thank you, I thought so. I will try with an ESP32 Version of the D1 Mini as soon as I get it (I already designed a PCB with an ethernet connector and it has a socket for the D1 Mini so I hope I can still use it xD)

pollierer, 2023-09-11:
can confirm: ESP8266 is not sufficient - works on an esp32

closed 2023-09-11 by pollierer