klatremis/esphome-for-deye #20: Total PV Power

Adminius, 2023-09-25:
Hi, Deye sends PV1 and PV2 separated only. Sometimes it's better to have total PV power already calculated

The only solution I could find until now:

```
sensor:
  - platform: template
    name: "${device_type}-Total PV Power"
    id: ${device_type}_Total_PV_Power
    unit_of_measurement: "W"
    state_class: "measurement"
    accuracy_decimals: 0
    lambda: |-
      return (id(${device_type}_PV1_Power).state + id(${device_type}_PV2_Power).state);
    update_interval: 5sec
```

But it's not synced (=> delayed) with PV1 and PV2 power, because it's not triggered by PV1 or PV2.
Is there any better solution for it?

MyHekla, 2023-09-26:
I have a similar calculation - but don't use the 'updateinterval'

  - sensor:
      - name: dsmr_grid_total
        unique_id: 1001
        unit_of_measurement: 'W'
        state: "{{ states('sensor.dsmr_reader_power_consumed') |float - states('sensor.dsmr_reader_power_returned') | float }}"
      - name: power_consumed
        unique_id: 1002
        unit_of_measurement: 'kWh'
        state: "{{ states('sensor.deye12k_sun12k_daily_production') |float +
                    states('sensor.deye12k_sun12k_daily_energy_bought') | float -
                    states('sensor.deye12k_sun12k_daily_energy_sold') | float}}"

Adminius, 2023-09-26:
Hm, now I'm testing this solution:

```
  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${device_type}-PV1 Power"
    id: ${device_type}_PV1_Power
    register_type: holding
    address: 672
    unit_of_measurement: "W"
    state_class: "measurement"
    accuracy_decimals: 0
    value_type: U_WORD
    on_value:
      then:
        - component.update: ${device_type}_Total_PV_Power
 
  - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${device_type}-PV2 Power"
    id: ${device_type}_PV2_Power
    register_type: holding
    address: 673
    unit_of_measurement: "W"
    state_class: "measurement"
    accuracy_decimals: 0
    value_type: U_WORD
    on_value:
      then:
        - component.update: ${device_type}_Total_PV_Power

  - platform: template
    name: "${device_type}-Total PV Power"
    id: ${device_type}_Total_PV_Power
    unit_of_measurement: "W"
    state_class: "measurement"
    accuracy_decimals: 0
    lambda: |-
      return (id(${device_type}_PV1_Power).state + id(${device_type}_PV2_Power).state);
```

total_pv_power should be updated each time PV1 or PV2 gets new value. let's see

Adminius, 2023-10-10:
my solution from previous post works pretty well, the only thing I changed: "on_value" only for PV2, because PV1 und PV2 are updated at the same time, it doesn't make sense to calucalate ithe sum two times.

closed 2023-10-10 by Adminius