klatremis/esphome-for-deye #21: no response received

dj9717, 2023-11-04:
hi,

i have:
[WT32-ETH01](https://www.amazon.de/dp/B09MD88B1N?psc=1&ref=ppx_yo2ov_dt_b_product_details)
[RS485 to TTL](https://www.amazon.de/dp/B083943T9J?psc=1&ref=ppx_yo2ov_dt_b_product_details)
[power supply](https://www.amazon.de/dp/B07C53B1GZ?ref=ppx_yo2ov_dt_b_product_details&th=1)
deye SUN-10K-SG04LP3-EU
I connect:
WT32 IO14 to TTL TX 
WT32 IO35 to TTL RX
WT32 3v3 to TTL VCC
WT32 GND to TTL GND
RS485 B+ to deye ModBus RJ45 1st Contact from the left
RS485 A+ to deye ModBus RJ45 2. Contact from the left
RS485 GND to deye ModBus RJ45 3. Contact from the left

The WT32 is available ( via Webservice ), but I don't get answers from the deye sun-10k.
i just get:
[22:10:22][C][modbus_controller:298]: ModbusController:
[22:10:22][C][modbus_controller:299]:   Address: 0x01
[22:10:24][D][modbus_controller:043]: Modbus command to device=1 register=0x271 countdown=0 no response received - removed from send queue
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=148 count=12
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=541 count=1
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=548 count=1
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=638 count=1
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=653 count=1
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=655 count=1
[22:10:25][W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=672 count=2
[22:10:25][W][component:214]: Component modbus_controller took a long time for an operation (0.06 s).
[22:10:25][W][component:215]: Components should block for at most 20-30ms.
[22:10:25][D][modbus_controller:043]: Modbus command to device=1 register=0x28F countdown=0 no response received - removed from send queue
[22:10:26][D][modbus_controller:043]: Modbus command to device=1 register=0x2A0 countdown=0 no response received - removed from send queue
[22:10:28][D][modbus_controller:043]: Modbus command to device=1 register=0x94 countdown=0 no response received - removed from send queue
[22:10:29][D][modbus_controller:043]: Modbus command to device=1 register=0x21D countdown=0 no response received - removed from send queue

and every now and then: 
22:38:20 | [E] | [api:128] | No client connected to API. Rebooting...

Does anyone of you have an idea what I'm doing wrong and can help me.

thanks

MyHekla, 2023-11-05:
I don't think your power supply is powerfull enough.

dj9717, 2023-11-06:
> I don't think your power supply is powerfull enough.
I have now used this [USB adapter](https://www.amazon.de/dp/B08T22HC8K?ref=ppx_yo2ov_dt_b_product_details&th=1) for power supply. Unfortunately the behavior doesn't change. how much power is required for stable function ?

What is also different here than described is that when I use the BMS port of the deye, only the TX LED of the RS485-TTL flashes evenly every second, but the RX LED of the RS485-TTL never flashes. If I use the Modbus port of the deye then the TX flashes again in time but also the RX flashes every now and then in short series. Unfortunately, I only see the unanswered messages in the WT32 log.

Alphaemef, 2023-11-09:
The BMS port is as far as I can tell, only read access to the modbus registry address. 

For safe measure, in your ESPhome configuration are you absolutely sure you have applied the correct Tx and Rx targets (the correct GPIOs?) On some TTLs the Rx and Tx have to be swapped. Did you try that ?

dj9717, 2023-11-11:
> The BMS port is as far as I can tell, only read access to the modbus registry address.
> 
> For safe measure, in your ESPhome configuration are you absolutely sure you have applied the correct Tx and Rx targets (the correct GPIOs?) On some TTLs the Rx and Tx have to be swapped. As in you keep your code, but just switch the wires to see if it changes anything.

yes, I have already swapped tx and rx.
If I do this, the RX LED of the RS485-TTL flashes in time and no longer the TX LED. The TX LED then remains completely off. The ESP still doesn't receive any replies from the WR.
Or maybe the inverter is the problem. I can test it in another way and as simply as possible.

klatremis, 2023-11-14:
> > The BMS port is as far as I can tell, only read access to the modbus registry address.
> > For safe measure, in your ESPhome configuration are you absolutely sure you have applied the correct Tx and Rx targets (the correct GPIOs?) On some TTLs the Rx and Tx have to be swapped. As in you keep your code, but just switch the wires to see if it changes anything.
> 
> yes, I have already swapped tx and rx. If I do this, the RX LED of the RS485-TTL flashes in time and no longer the TX LED. The TX LED then remains completely off. The ESP still doesn't receive any replies from the WR. Or maybe the inverter is the problem ? Is there a way to test it differently and as easily as possible ?

I dont think its a inverter problem.
I belive its a wiring issue. 
1. Make sure B go to B, A go to A, GND to GND on the between TTL board and inverters "modebus" port.
2. Use the correct RX/TX pins, and swap rx/tx if it dosent respond correct.

It could also be that the ESP32 version you use is using the rx/tx for the lan port maybe?? Try another type of esp32 without LAN port.

andriymoroz, 2025-01-06:
Few things to check for people debugging similar failure ("no response received")
- rx on ESP connected to RX to UART-RS485 module and TX to TX
- when assembled and powered before connecting to the inverter, TX led on the converter should blink periodically
- on the inverter go to Advanced settings, page 2 and check the modbus address of your unit. Should be the same you specified in the `modbus_controller: / address: XXX` section of the ESPHome device config. Change in either place to match

malcolmx67, 2025-06-14:
Hi,
I got the same problem meaning tx LED on RS485 board is blinking, bot nothing gets received. 
My inverter has modbus address 0 (slave), so I also changed the esp config to this address. However I thought, that the address within the yaml was supposed to be the address of the esp32 (not the inverter address). Anyhow, still doesn't work. I triple-checked all connections, tried different Rx / Tx pins w/o success.
Can anybody confirm, that the modbus address defined within the yaml 

modbus_controller:
  - id: ${modbus_controller_id}
    address: 0x1

represents the inverter address and therefore has the match the configured one ?
BTW: 
Switching Rx / Tx doesn't help either ....

TIA

MyHekla, 2025-06-14:
that is working in my config.

goloveychuk, 2026-01-08:
Idk if it’s faulty converter or I messed with soldering it, but I have same no response issue. Then I bought rs232 adapter, connected to wifi port - and it worked from first attempt, no firmware changes. 
Additionally, wifi port has 12v on 9 and 5 pins, so you don’t need external power. 
Aliexpress has converters with db9 port. 

[image]
[image]
[image]