klatremis/esphome-for-deye #22: 98 - Control Mode - Enable/Disable Battery?

TheSmartGerman, 2023-11-07:
Hello,
I need some thoughts about this: is it possible to enable / disable the battery remote?

I found the "98 - Control Mode" but it's only 0 and 1 for leadascid and lithium in the register discription.

Is it possible to set also 2 or 3? I can't finde any explenation...

if the Control Mode is the right address, it should work like this?

```
- platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: ${device_type}-control-mode
    address: 98
    value_type: U_WORD
    optionsmap:
      "Lithium": 0
      "Use Bat V": 1
      "Use Bat %": 2
      "No Bat": 3
```

[image]

fabhund, 2023-11-07:
Not sure if Addr: 98 or 111 can do it.
But if you diable the battery on the control panel the inverter trips, and gives an alarm. And may need to be reset before it connects again. You man risk doing the same here over ModBus

Other methods to "disable" the battey is to set the charge / discharce ampere to 0A when needed.
Or change the "Time of use" table. If fx. you dont want any discharge, set the capacity to 100%

TheSmartGerman, 2023-11-07:
ok, well I can read the registers and monitor the changes, while changing them by hand.

Yes, if you change to "no battery" there will be a error. If I remember correctly it's something about "Grid mode change". A Relais is switching -> there is no restart neccessary. It will switch to "off" in the middle circle after 1 minute or so. 

Sure, you can stop charging/discharging of the battery. But as long as the battery is configured the inverter needs much more standby power.

For winter / bad weather periodes, It's much better to set the inverter to "no Batt", you can save ~ 2kWh a day.

fabhund, 2023-11-07:
> For winter / bad weather periodes, It's much better to set the inverter to "no Batt", you can save ~ 2kWh a day.

Ahh, Yes. By having "No Batt" the inverter will actually shut down. And you consume from grid in cheaper (night) periods.
And Yes, maybe monitor the registers in question and make the change. (I would like to know the result, as I'm having 2 inverters, and at night time I could turn 1 off.

TheSmartGerman, 2023-11-08:
I did the test, you need both registeres:
98 - 0 / 111 - 0 -> Lithium (BMS)
98 - 1 / 111 - 1 -> Use V
98 - 1 / 111 - 2 -> Use %
98 - 1 / 111 - 3 -> No Bat

It works fine. See code below. Now I have another question: Is it possible to build a Select  menu and manipulating both registeres?

```
select:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "${device_type} Control Mode"
    id: ${device_type}_control_mode
    address: 98
    value_type: U_WORD
    optionsmap:
      "Lithium Battery": 0
      "Lead Battery": 1

  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "${device_type} Battery Operation"
    id: ${device_type}_battery_operation
    address: 111
    value_type: U_WORD
    optionsmap:
      "Use V": 0
      "Use %": 1
      "No Bat": 2
```

Alphaemef, 2023-11-08:
> I did the test, you need both registeres: 98 - 0 / 111 - 0 -> Lithium (BMS) 98 - 1 / 111 - 0 -> Use V 98 - 1 / 111 - 1 -> Use % 98 - 1?0 / 111 - 2 -> No Bat -> Grid Change Warning (F13) will occure.
> 
> For "No Bat" I saw 0 or 1 for register 98. I guess it's ignored. Just depents on the previous settings. If you jump from Lithium to "No Bat", it will stay at lithium. If your used "User V / Use %" for user definded settings, it will stay at 0 "Lead Battery".
> 
> It works fine. See code below. Now I have another question: Is it possible to build a Select menu to manipulating both registeres?
> ```

I think there is an error in your mapping. This is not what I have form my testing. 
Try: 
`
select:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} Battery Control Mode"
    id: batt_control
    address: 111
    value_type: U_WORD
    optionsmap:
      "Voltage Mode": 0
      "SoC mode": 1
      "No Battery": 2
    skip_updates: ${skip_updates1439}
    optimistic: true
    disabled_by_default: true
    on_value:
      then:
        - if:
            condition:
              lambda: |-
                return id(batt_control).state == "Voltage Mode";
            then:
               - switch.turn_on: lithium_mode

  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "${name} Battery - Lithium Mode"
    id: lithium_mode
    register_type: holding
    address: 98
    bitmask: 1
    icon: "mdi:toggle-switch"
    on_turn_on: 
      lambda: |-
        auto call = id(batt_control).make_call();
        call.set_option("SoC mode");
        call.perform() ;

`
Let me know if it works :)

Alphaemef, 2023-11-09:
> I did the test, you need both registeres: 98 - 0 / 111 - 0 -> Lithium (BMS) 98 - 1 / 111 - 0 -> Use V 98 - 1 / 111 - 1 -> Use % 98 - 1?0 / 111 - 2 -> No Bat -> Grid Change Warning (F13) will [...]

I think there is an error in your mapping. This is not what I have form my testing. 
Try: 

```
select:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "Battery Control Mode"
    id: batt_control
    address: 111
    value_type: U_WORD
    optionsmap:
      "Voltage Mode": 0
      "SoC mode": 1
      "No Battery": 2
    optimistic: true
    disabled_by_default: true
    on_value:
      then:
        - if:
            condition:
              lambda: |-
                return id(batt_control).state == "Voltage Mode";
            then:
               - switch.turn_on: lithium_mode
switch:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "Battery - Lithium Mode"
    id: lithium_mode
    register_type: holding
    address: 98
    bitmask: 1
    icon: "mdi:toggle-switch"
    on_turn_on: 
      lambda: |-
        auto call = id(batt_control).make_call();
        call.set_option("SoC mode");
        call.perform() ;
```

Let me know if it works :)

TheSmartGerman, 2023-11-09:
@Alphaemef You are right, i mixed the battery type (Control Mode):

[image]

Therefore it should be: 
```
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "${device_type} Control Mode"
    id: ${device_type}_control_mode
    address: 98
    value_type: U_WORD
    optionsmap:
      "Lead Battery": 0
      "Lithium Battery": 1
```
I should be: !! Not testet !!
```
98 - 1 / 111 - 0 -> Lithium (BMS)
98 - 0 / 111 - 0 -> Use V
98 - 0 / 111 - 1 -> Use %
98 - 0 / 111 - 2 -> No Bat -> Grid Change Warning (F13) will occure.
```
I'll test your code the next few days, I should take more time to be sure not to do more mistakes :(

pvprodk, 2023-11-14:
> ```
> select:
>   - platform: modbus_controller
>     use_write_multiple: true
>     modbus_controller_id: ${modbus_controller_id}
>     name: "Battery Control Mode"
>     id: batt_control
>     address: 111 [...]

I have testet this.
When I set address 111 to 2 (No Battery) the inverter goes to fault, and after a short while to standby
When I set address 111 to 1 (Soc Mode) the inverter resumes to normal operation (Lithium BMS controlled)

While the 111 is set to 2 (No battery) the display on the inverter still shows "Lithium" selected, but I dont care as long as it works. So in my case I dont even use the address 98 part.

pvprodk, 2023-11-14:
BTW, I am running firmware version 1140 on the Deye inverter

klatremis, 2023-11-14:
> > ```
> > select:
> >   - platform: modbus_controller
> >     use_write_multiple: true
> >     modbus_controller_id: ${modbus_controller_id}
> >     name: "Battery Control Mode"
> >     id: batt_control
> >     address: 111
> >     value_type: U_WORD
> >     optionsmap:
> >       "Voltage Mode": 0
> >       "SoC mode": 1
> >       "No Battery": 2
> >     optimistic: true
> >     disabled_by_default: true
> >     on_value:
> >       then:
> >         - if:
> >             condition:
> >               lambda: |-
> >                 return id(batt_control).state == "Voltage Mode";
> >             then:
> >                - switch.turn_on: lithium_mode
> > switch:
> >   - platform: modbus_controller
> >     use_write_multiple: true
> >     modbus_controller_id: ${modbus_controller_id}
> >     name: "Battery - Lithium Mode"
> >     id: lithium_mode
> >     register_type: holding
> >     address: 98
> >     bitmask: 1
> >     icon: "mdi:toggle-switch"
> >     on_turn_on: 
> >       lambda: |-
> >         auto call = id(batt_control).make_call();
> >         call.set_option("SoC mode");
> >         call.perform() ;
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Let me know if it works :)
> 
> I have testet this. When I set address 111 to 2 (No Battery) the inverter goes to fault, and after a short while to standby, as intended. When I set address 111 to 1 (Soc Mode) the inverter resumes to normal operation (Lithium BMS controlled)
> 
> While the 111 is set to 2 (No battery) the display on the inverter still shows "Lithium" selected, but I dont care as long as it works. So in my case I dont even use the address 98 part. Address 98 is set to 1 all the time.
> 
> [image]

Cool. I think i will try play with this, and see myself how it react on it.

vejvik, 2024-03-29:
please, which ESP code is right for me if I want to be able to switch battery mod - SoC and voltage options (I don't want no battery)
Thanks