klatremis/esphome-for-deye #24: Daily Consumption

tobenski, 2023-11-14:
Hi There, 

I was trying to create a Daily consumption sensor, and i believe the modbus adress is 526 but it seams the messurment is of by quite a bit. (a least compared to the Solarman Logger interface)

My sensor code is this: 
```
 - platform: modbus_controller #Daily Consumption
    modbus_controller_id: ${modbus_controller_id}
    skip_updates: 10
    name: ${device_type}-Daily Consumption
    id: ${device_type}_Daily_consumption
    register_type: holding
    address: 526
    unit_of_measurement: "kWh"
    state_class: "total_increasing"
    device_class: energy
    accuracy_decimals: 2
    value_type: U_WORD
    filters:
      - multiply: 0.1
```
 
Have you tried getting this sensor to work?

klatremis, 2023-11-14:
I just use to the total, and let home assistant tell me how high the daily was. When using a daily sensor, its then the inverter that decide when the time reach 00. And not home assistant. Deye inverters tent to drift in clock over time. 5-10min/month have i seen. Therefor the daily is less accurate than a total and let HA controll the daily instead.

Without trying, i would just make it like i did for 527
How much is it off?

tobenski, 2023-11-15:
Hi again, 
I actually made an utility sensor for the daily production bassed on the "total_consumption".
It is showing the same messurement as the "Daily_consumption"

but both numbers ar eof by quite a bit.

Solarman:                    ESP
15.1

klatremis, 2023-11-22:
What does the energy dashboard in HA say?

If the solarman use a different timezone, it will be different.