klatremis/esphome-for-deye #25: Entities becomes unavailable

chinezbrun, 2023-11-21:
after several days of normal operation the entities in HA becomes unavailable.

Home Assistant 2023.10.5
ESPHome 2023.11.2

WiFi ok. module is online 
Reinstalled - no results
Delete device from HA - no results

log shows:
22:19:02][D][modbus_controller:043]: Modbus command to device=1 register=0x94 countdown=0 no response received - removed from send queue
[22:19:03][D][modbus_controller:043]: Modbus command to device=1 register=0xA6 countdown=0 no response received - removed from send queue
[22:19:04][D][modbus_controller:043]: Modbus command to device=1 register=0x154 countdown=0 no response received - removed from send queue
[22:19:06][D][modbus_controller:043]: Modbus command to device=1 register=0x21D countdown=0 no response received - removed from send queue
[22:19:07][D][modbus_controller:043]: Modbus command to device=1 register=0x224 countdown=0 no response received - removed from send queue

klatremis, 2023-11-22:
What hardware do you use?

This is normal feedback when the rs485 communication is not working.

chinezbrun, 2023-11-24:
Hardware is exactly what you recommended and the power is coming from pin7,8 .
The communication works, and suddenly become like this, and  this worry me.
I will try hardware restart with power plug.

fabhund, 2023-11-24:
For me, it has been a bit of hit'n'miss with the hardware. Meaning I went through multiple ESP32 and TTL<->RS485 boards. Also tried to jump the ground connection across the TTL board. I cannot give a reason other that try another ESP32 + TTL board. It's been working stable for me for quite a while. Also powerd from CN2 Pin7+8. Via a 5V USB stepdown module. I would assume a burnt out TTL module. I also reduced the update interval to 15s. Maybe check if  the 5V is stable. Maybe add a few filter capacitors to 5V +3.3V (Not sure it will help anything) Like I said. Hit''n'miss :-/

MyHekla, 2025-06-14:
please close this one