klatremis/esphome-for-deye #31: Set the Deye at its lowest possible self-consumption

Breinholm-coder, 2024-02-29:
Hej Klatremis,
I have 2 parallel 12 kW Deye hybrid inverters - like yours.
I want to reduce their power usage to the absolute minimum during night-time. They have a rather high idling power level (around 160W) and that's too high and ineffective compared to my house load of about 350 W. I have therefore invested in a microinverter that I crank up when our power usage is low and the sun-production has ended for the day.
I can see there is a Modbus 80 register that could be a software ON/OFF switch for the Deyes - Have you tried to address that register?
If using 80 is not feasible can you then suggest alternatives to reduce the Deye power level?
Thanks / Peter

pvprodk, 2024-02-29:
You can play around with theese modbus registers.
Setting the Battery Control Mode to "No Battery" reduces the idle consumption to ~15W
But the inverter goes into error-state for around 15 mins when going to "No Battery", but I guess that doesn't matter if its at night.
You could make automations to trigger "No Battery" and back to "SoC Mode" 

```
switch:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "Deye Battery - Lithium Mode"
    id: lithium_mode
    register_type: holding
    address: 98
    bitmask: 1
    icon: "mdi:toggle-switch"
    on_turn_on: 
      lambda: |-
        auto call = id(batt_control).make_call();
        call.set_option("SoC mode");
        call.perform() ;

select:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "Deye Battery Control Mode"
    id: batt_control
    address: 111
    value_type: U_WORD
    optionsmap:
      "Voltage Mode": 0
      "SoC mode": 1
      "No Battery": 2
    optimistic: true
    disabled_by_default: false
    on_value:
      then:
        - if:
            condition:
              lambda: |-
                return id(batt_control).state == "Voltage Mode";
            then:
               - switch.turn_on: lithium_mode
```

switschel, 2024-03-01:
Are you sure you need to set the 98 register? I'm just switching 111 register only every day from "No Battery" to "SoC mode" and vice versa and it is working very fine.

pvprodk, 2024-03-01:
> Are you sure you need to set the 98 register? I'm just switching 111 register only every day from "No Battery" to "SoC mode" and vice versa and it is working very fine.

Thats also what I am doing, but the cude in register 111 addresses the code in 98, so I have not dared to remove it ;)

kababoom, 2024-03-01:
Using the same, working good.

```
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: ${modbus_controller_id}
    name: "sun3p-Battery mode"
    id: sun3p_battery_mode
    address: 111
    value_type: U_WORD
    optionsmap:
      "BattV mode": 0
      "No Battery": 2
```

Breinholm-coder, 2024-03-01:
Thanks guys - I will try the 111 register.
Have you all also added a microinverter, connected to your batteries, for low-power supply?
It works flawlessly (When I depress the blue button manually though) and reduces the power drainage by about 250 W.

drakesoft, 2024-08-28:
The best solution for reducing self consumption is to modify the "blue" button with a relay. 
Than the consumption can be reduced to 25W. If the Grid is also switched with a power contractor than the inverter can be completely switched of. But than the power up takes a minute. 
Switching Grid without the blue button is no solution because than the inverter run in island mode.

Breinholm-coder, 2024-08-28:
I have used writing to register 80 (Inverter On/Off) on the Deye master master since I started this thread.
It works well - But the Deyes enters into error. The error is reset after turning them on again. Tried writing to register 98, but the power consumption was higher.