klatremis/esphome-for-deye #32: Microinverter on the generator port

Renejeppesen, 2024-03-19:
Hello

I have configured the genrator port for microinverter connection. 

How do I display the delivered displayed on this port?

philipphenkel, 2024-03-20:
Hi,

I'm reading the GEN port power from register 667, see [deye-sun-12k-sg04lp3.yaml](https://github.com/philipphenkel/esphome-config/blob/main/devices/deye-sun-12k-sg04lp3.yaml)

 ```
 - platform: modbus_controller
    modbus_controller_id: ${modbus_controller_id}
    name: "${entity_name_prefix}Gen Power"
    register_type: holding
    address: 667
    unit_of_measurement: "W"
    accuracy_decimals: 0
    device_class: power
    state_class: "measurement"
    value_type: S_WORD
    
```

Renejeppesen, 2024-03-24:
Thanks it worked

Renejeppesen, 2024-03-24:
Thanks it worked

closed 2024-03-24 by Renejeppesen