klatremis/esphome-for-deye #38: Erratic inputs from Deye 12K repeatedly received

parkflyer, 2024-05-03:
I am using this integration now for several month on my Deye 12K inverter and have observed that received values for different entities (sun12k-pv2_power, sun12k-battery_capacity and a lot more, but not all of them) to be completely out of range, e.g. sun12k-pv2_power being set to 30806kW (real maximum should be around 10.000kW). It seems to appear at random and only for a short period of time.
During this "condition" some entities seem to be set to fixed values (e.g. "30806" appears on several other entities), maybe depending on data type (integer, float, ...) or units.
[image]
[image]

The behavior started to appear on April 23rd, see included picture (peaks = wrong values). The effect seems to get stronger with higher solar radiation / higher values.
From the Debug log:
```
2024-05-01 11:20:26.596 DEBUG (MainThread) [aioesphomeapi.connection] deye-modbus @ 192.168.0.212: Got message of type SensorStateResponse: key: 3933863750
state: 4114 (correct)
... (other entities)
2024-05-01 11:21:17.823 DEBUG (MainThread) [aioesphomeapi.connection] deye-modbus @ 192.168.0.212: Got message of type SensorStateResponse: key: 3933863750
state: 30806 (incorrect)
... (other entities)
2024-05-01 11:36:54.946 DEBUG (MainThread) [aioesphomeapi.connection] deye-modbus @ 192.168.0.212: Got message of type SensorStateResponse: key: 3933863750
state: 3749 (correct)
```

I have no clue but it somehow looks that with certain values / bit combinations the reception of data is not working correctly.
Any ideas?

Thank you,
Ewald

parkflyer, 2024-05-03:
Additional info: I cannot see these peaks in the HA Solarman integration, which is running in parallel (using the Deye WLAN connection).

parkflyer, 2024-05-04:
Hi!

Sample rate should be default (15 sec) - I did not change it and 15 sec match the update interval I see in HA.

I am not using shielded cables, but the connection between inverter and RS485 module is only 50 cm and between module and ESP32 only a few cm).
And there were no problems before April 23rd using the same hardware (I've added a third 5kW battery a few days before and could have touched the module and cables, but normally this should have no impact).

About the isloated RS485 transceiver I'm not sure, I think it is not isolated (looks like the one in the esphome-for-deye README).

Today again was very bad:
[image]

Thanks,
Ewald

luckylinux, 2024-05-04:
Did you try rebooting the ESP32 ?

I had weird issues happening with log level set to DEBUG, so it could be that after a while some issues pop up. Try setting to INFO.

About your 30806 it's a bit weird. If it was 32767 I would have told you it's probably an integer overflow for a 16-bit SIGNED Integer (16 bit signed is between -32768 and +32767).

Did you try to increase your Stack Size ?

See an example here:
https://github.com/luckylinux/esphome-for-deye/blob/main/boards/esp32-s3-wroom-1-n16r8.yaml

I was doing a lot of troubleshooting for a very separate issue (OTA) so the file is not clean at all. But try to implement the part for the option:
```
- -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768
```

It's been running OK for me for approx. 1 Month on an M5Stack Atom S3 Lite with Isolated RS485 Transceiver.

parkflyer, 2024-05-07:
Thank you very much for your quick and comprehensive support!
I have started with the easy fixes and attached a USB power adapter (battery based) as power supply - hooray, no more spikes! 
So I reattached the Deye 12V supply again to see if was just the restart - no luck, spikes reappeared.
My next attempt was to add a 330uF 35V low-ESR capacitor between +5V and GND on the ESP32 - again, no luck.
So I‘m now adding a power outlet next to the inverter and will use a USB PSU to feed the device.
Thanks again for your help!

luckylinux, 2024-05-07:
No worries. 

I'm sure there were many other Issues reported that required some capacitor etc or a different power supply than the CN2 pins (you might want to browse the "Issues" Section including the closed ones if you want to go down that road, I cannot remember if it was on "both" sides of the Buck Converter or only on the Output).

I suspect that the CN2 Power is not so stable / stabilized. So maybe Inverter Load Response coupled with your Buck Converter causes some Power Supply instability. 

Or it could also be EMC related.

Otherwise yeah ... Just use an external PSU and be happy :smile:.

You can also do it the other way if it's preferable: I used a 10m long CAT6 Cable so my ESP32 sits in the Control Cabinet, away from the Inverter, and close to a 230VAC Socket.

Steiner606, 2024-06-19:
Hi, I've got the same issue. It seems only to appear when the Converter has a high load. It was running since Sebtember last year without any issues.  Since a month or so some measurements show the described high values for a very short time. In particular it`s the battery capacity and the PV Power. I'll try to use a separate power supply as well. I hope thats gonna help.

maskedfighter, 2024-11-05:
I do have the same problem with spikes, i am reading out the deye witha lilygo rs485 poe board which is powered by ethernet power alone, setup is very clean cause of less cables but i wonder how those spikes appear... did anyone try getting rid of the esp32 integration and use f.e. waveshare poe or similar? And on which port on the deye did you guys have your rj45 cable plugged in for reading out modbus?
I mean, its just a slight inconvenience to see the spikes but it just tickles my mind that it doesnt work as it should...
I also read that there is a sunsynk asdon for HA which is also able to read out data waveshare devices provide... but i wonder if waveshare devices then also have those spikes shown?
I don't see any spikes when using the wlan dongle and solarman integration... (but i want to have a proper thethered connection to my inverter).

Steiner606, 2024-11-06:
Hi, I had the same issues (see the post before) but since I'm using a separate Power supply for my ESP32 board the spikes are gone and it works like a charm. I'm using a normal ESP32 board with USB Power supply.

maskedfighter, 2024-11-06:
Yep, i already installed a 12V din rail power supply, thank god i installed the esp case next to the power distribution box. 
Will try it out later, thx for the input! :) 
Btw, i set logger baudrate to 0, far less spikes since then... 
PoE ESP Boards would be so convenient if they just worked as intended... :(

maskedfighter, 2024-11-22:
Same issues with external 3A power supply (24V)... No idea...

fabhund, 2024-11-23:
I had the same issue, also had the TTL board "locking up" I ended up trying multiple ESP32 and TTL boards. Now i have a 5V Buckconverter, ESP32 and TTL board all placed inside the inverter. (With the screw terminals) So maybe order 5-10 different boards, and start replacing....

maskedfighter, 2024-12-06:
Found the reason that caused the spikes. It is the included rs232 wifi dongle. I had it running aloneside with this plugin, and guess what, as soon as i inplugged the dongle, there are absolutely 0 spikes recorded anymore. Thank god it was that easy 😇

MyHekla, 2025-06-14:
please close this issue