klatremis/esphome-for-deye #41: Daily charge of battery

jais86, 2024-07-03:
Hi,

Is it possible to add a daily charge and discharge of battery?

fabhund, 2024-07-03:
No. To my knowledge there is no Modus register that has that value from the inverter itself.
Assuming you want this in HomeAssistant, then the energydashboard calculates this for you on a hourley/daily/weekly/monthly/quaterly/yearly basis anyway.

Baduno, 2024-08-20:
What sensor do you use under "Energy - Home battery storage" to make this worke?

luckylinux, 2024-08-20:
I'm not sure if it's being asked to CHARGE or DISCHARGE (like "Actually do it") or just what Amount of Charge/Discharge was done Today or in Total (like "Monitor").

For the latter, you can try
[image]

No guarantees, I did NOT test this

luckylinux, 2024-08-20:
Actually I was using it for the "Total Charge/Discharge":
https://github.com/luckylinux/esphome-for-deye/blob/main/modules/battery.yaml

So you just would need to adapt that to registers 514 and 515 I guess.