klatremis/esphome-for-deye #43: Functionality request

rvdgaag, 2024-09-10:
Hi I would like to have a button to switch off in total the Deye inverter, I believe modbus register 80 makes this possible ?

How do I add this to the yaml ?

fabhund, 2024-09-11:
Please note this will not turn off he inverter _completly_ it will only stop the inverter/charging function. The display etc. will still be operational.
Further it throws a F19, that is rather undocumented, so proceed with caution. Fx. if you turn it off, and for some reason you loose connection via.modbus you will not be able to turn it on via. the Display again.

Add this to your "Switch" segment in the YAML, and edit name, modbus ID etc. to your needs.
https://pastebin.com/mYznfnfg

switch:
  - platform: modbus_controller
    use_write_multiple: true
    modbus_controller_id: Put your modbus_controller_id here
    name: Deye12k_Power_switch
    register_type: holding
    address: 80
    bitmask: 1
    entity_category: config

fabhund, 2024-09-12:
It should. The inverter trips and stops all inverting. But test it out for yourself, again there is very little documentation for what it does, and throwing an error is also wierd.
As an alternative you could re-wire the on/off switch of the inverter to a external Relay controlled via ESP32 (ESPhome) this way you can use the existing built-in function to start/stop the inverter. 2.nd alternative is to mount a "Switchbot" on top of the button to physically push the button.

b1ggi, 2024-11-01:
I have no Load/Gen/PV connected to my deye 12k, here are my findings:
i used the register 80 to turn off, it worked well until the Battery went into UVP (no problem, enough left). 
- The CAN Communication did turn off according to YamBMS. 
- The Capacitor for the Battery discharged from 44.15v to 9v because the BMS relay kicked in

[image]

Later i wanted to turn on the through register 80, but only the Grid Inverter switched on. the inverter relay stayed off. I had to press the blue Button on the left to charge the capacitor again.