klatremis/esphome-for-deye #46: Info esp32+m5stack-atom

wifi75, 2025-04-03:
is it possible to use this combination esp32+m5stack-atom to control deye inverter?

MyHekla, 2025-04-23:
requirements is described on the main page - you also need RS485.
What do you expect from the M5?

wifi75, 2025-04-23:
> requirements is described on the main page - you also need RS485. What do you expect from the M5?

M5STACK ATOM Lite + M5STACK ATOM RS-485 Base

[image]

[image]

MyHekla, 2025-04-23:
I asked what YOU expected!

wifi75, 2025-04-23:
> I asked what YOU expected!

basically i just have to change the correct rx and tx part number?