klatremis/esphome-for-deye #47: SUN-10K-SG04LP3

tmbreuker, 2025-05-20:
Hi,

I stumbled upon this in search of a solution to manage my SUN-10K-SG04LP3 through Home Assistant. I used a similar setup with my heatpump. My problem is: I can wire up hardware and setup home assistant, but getting the correct hardware and pogramming esphome is something way out of my comfort zone. Is it possible to get some assitance here, eg. what hardware to get and how to connect it to my Deye SUN-10K-SG04LP3? 

Thanks,
Tom

Emacee, 2025-05-22:
That device won’t work with this software. You will need an ESP32 based device together with RS485 communication. You can buy these components separately but then you will need to solder them together to get it to work.

Something [like](https://lilygo.cc/products/t-can485) this should work (not tested myself though);

- ESP32 based
- RS485 module built in
- Power In voltage in line with what the Deye inverter can provide

luckylinux, 2025-05-22:
M5Stack ? No soldering required 😃 

- Base Board: ATOM S3 LITE C124 (with ESP32-S3 SOC): https://www.digikey.dk/en/products/detail/m5stack-technology-co-ltd/C124/18070571

And one of these Options (note that the GPIO Pins are different depending on which one you use):
- RS485 Module (Option A): ATOM U094 https://www.digikey.dk/en/products/detail/m5stack-technology-co-ltd/U094/13913913 
- RS485 Atomic Base (Option B): A131 https://eu.mouser.com/ProductDetail/M5Stack/A131?qs=Imq1NPwxi75XShTuCGBiEA%3D%3D

Optional if you don't want to cut your RJ45 Cable: https://www.digikey.dk/en/products/detail/dfrobot/FIT0849/15848077

tmbreuker, 2025-05-23:
Thanks @Emacee and @luckylinux As I am in the Netherlands, I go with option suggested by @luckylinux Would you be able to help me what power supply fits for the ATOM S3? The datasheet is not clear on what powersupply connector is used (mini USB or else?).

luckylinux, 2025-05-23:
@tmbreuker: https://docs.m5stack.com/en/core/AtomS3%20Lite#specifications

It's a USB-C Connector. Power Supply is standard USB 5V (Passive). You do NOT need USB-PD (that is typically used for higher Voltages e.g. 9V, 12V, 15V, ...). It **should** be backwards compatible and default to 5V if the Device doesn't negotiate a higher Voltage.

As for which Power Supply you could use something like
https://www.amazon.de/-/en/Raspberry-Pi-official-power-supply/dp/B07TMPC9FG

But as I said, that is quite overkill ...

For the Deye, I personally am using a 4 Port USB Power Supply (5V) that I was previously using for some Phones in the Past. For the BMS Monitoring on the other Hand I just put a Raspberry Pi PSU for each since it's more Critical (losing an Inverter **Monitoring** isn't the End of the World, losing the BMS will make the Inverter Trip since `BMS_Err_Stop` is enabled).

Just make sure it's some Good Power Supply and be Happy 😃.

You can try to save a few Euros here and there (or connect to CN2 Port of Deye and, if/when Issues arise, spend the next few Days try to troubleshoot a Problem that doesn't exist on the ESP32 but rather seems to be due to some very Limited Power Rating of the Deye CN2 Port combined with some Transients). For 10-15 EUR (minus the DC/DC which you would need to Buy if you use the CN2 Port) I though it just wasn't worth it ...

tmbreuker, 2025-05-23:
@luckylinux is there a manual how to connect all this to my deye, preferably including images?

luckylinux, 2025-05-23:
@tmbreuker: Look at the Bottom of my Repository README:

https://github.com/luckylinux/esphome-for-deye?tab=readme-ov-file#hardware-diagram

That's the Pinout of the Connector both for their Female RJ45 (Socket) and the Male RJ45 (Plug that you come with and attach to the Deye MODBUS Port).

If you want the Quick & Lazy Solution I suggest you but a Cat 5e/Cat6 (Beware that the Shielded Cable can be quite Hard to bend, yet I think it would be better to use a shielded Cable !) and use the Breakout Adapter I mentioned above (or something Similar from Aliexpress etc):
https://www.digikey.dk/en/products/detail/dfrobot/FIT0849/15848077

Then you will need 3 short wires to bridge between that Breakout Adapter and the RS485 Module and a small Screwdriver.

I cannot tell you Pin 1 ... 8, Left Right, because that depends on WHICH Connector (Male or Female), HOW you hold the Connector (Locking Tab up or down ?), etc.

luckylinux, 2025-05-26:
> I've got the hardware now but too stupid to read the hardware diagram
> Here the components I got: [image]

As I said I cannot really tell you Left/Right etc as it depends how you hold your RJ45 Adapter basically and which Plug (Female, Male) it is.

[image]

So:
- The BLUE (B) Terminal connects to Pin 1 (or Pin 8)
- The GREEN (A) Terminal connects to Pin 2 (or Pin 7)
- The BLACK (G = Ground) Terminal connects to Pin 6 (or Pin 6)

Furthermore, place the Resistor between Terminal BLUE (B) and Terminal GREEN (A), taking Care NOT to cause any short Circuit. You may shorten & cut the Pads a bit if they are too long, or alternatively heat-shrink them. It shouldn't cause any dangerous Situation if it shorts, but you will not have Communication in that case.

So actually you cannot do it wrong with this one. I had to deal with the CAN for the BMS (another Port) and that one is NOT foolproof 🤣.

> , also the setup steps are:
> 
>     1. Create your esp32 in esphome in home assistant > it seems I first need to setup my atom S3 - as ESPHome asks me for hostname at setup
> 
>     2. Upload the your basis config via. usb from pc. > this can be done with esphome, but I'm caught in chicken-egg situation
> 

It's just the usual Procedure really.

1. Clone Repository of Project
2. Change Configuration
3. Compile (`esphome build`)
4. Upload (`esphome run`)
5. Register your ESPHome Device into Home Assistant

This Script does most of the Stuff for you:
https://github.com/luckylinux/esphome-for-deye/blob/main/build.sh

Of course you need to copy `secrets.sh.example` to `secrets.sh` and modify both that and `devices.sh` according to your Setup.

Or just use the standard ESPHome Python Program, but then each Time you will have to remember what to do etc.

If you are on Windows you'll have to ask somebody else. I have no Experience there with ESPHome. You should be able to get it working from WSL2 Distribution or using a Virtual Machine e.g. Ubuntu inside Virtualbox or similar.

luckylinux, 2025-05-26:
If that doesn't work I can only suggest you spin up a Linux Virtual Machine with Ubuntu or something like that and pass-through the USB Adapter Matching the USB Port (or the entire USB Controller if you can do that without being "locked out" with Keyboard & Mouse), then try it from there. Or a LiveUSB with Ubuntu on it might also be an Option.

On Ubuntu AMD64 it works out of the Box.

I have no experience on flashing the ESP32-S3 from a AARCH64 (?) Machine though.

luckylinux, 2025-05-27:
@tmbreuker: Yeah, of course the Configuration needs to be adjusted.

Make sure you do **NOT** delete the Master though otherwise you will NOT have (almost) any controllability. Just delete the Slaves Devices 😉

tmbreuker, 2025-05-27:
@luckylinux I tried building but it ends with:
INFO ESPHome 2025.5.1
INFO Reading configuration esphome-config-.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Error reading file esphome-config-.yaml: [Errno 2] No such file or directory: 'esphome-config-.yaml'
In case of errors, it is suggested to run: esphome clean.

I guess my mac misses something...

luckylinux, 2025-05-27:
@tmbreuker: more like you deleted / unset the `profile` Variable in `devices.sh`:
https://github.com/luckylinux/esphome-for-deye/blob/main/devices.sh#L37

You can also change completely and do everything in one File, it's up to you.

But if you change Stuff in one Place, you must also change it in other Places ...

tmbreuker, 2025-05-27:
@luckylinux no, i merely took out the S02 and S03 parameters, using a # :
[image]

luckylinux, 2025-05-28:
> I am stuck here:
> 
> I tried running it on my synology. This time I get slightly further (after manually copying secrets.yaml in each subfolder config, modules, etc):

That should NOT be required really ...

> INFO Reading configuration esphome-config-master.yaml... Failed config
> 
> captive_portal: [source config/captive_portal.yaml:1]
> 
> This feature is only available with frameworks ['arduino']. {} ota: [source config/ota.yaml:2]

That's weird, I didn't have that before.

I tried now to upgrade to latest ESPHome.

Currently I'm running (I never upgraded before, it if works, don't touch it 😛):
```
Firmware: 20240820 (ESPHome 2024.7.3)
Hardware: 20240820 
```

And
```
Firmware: 20240405 (ESPHome 2024.3.1)
Hardware: 20240405 
```

However, just to prove my Point, I built it for my 3rd Device (`deye-c`) and it built without an Issue:

```
uilding .pioenvs/deye-c/bootloader.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
Linking .pioenvs/deye-c/firmware.elf
RAM:   [=         ]  11.0% (used 36108 bytes from 327680 bytes)
Flash: [======    ]  57.7% (used 1058225 bytes from 1835008 bytes)
Building .pioenvs/deye-c/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/deye-c/firmware.bin"], [".pioenvs/deye-c/firmware.elf"])
SHA digest in image updated
Wrote 0x112720 bytes to file /home/USER/ESPHome/esphome-for-deye/.esphome/build/deye-c/.pioenvs/deye-c/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/deye-c/firmware.bin"], [".pioenvs/deye-c/firmware.elf"])
=============================================================================================================== [SUCCESS] Took 150.32 seconds ===============================================================================================================

**************************************************************************************************************************************************************************************************************************************************************
We found 504.96MB of unnecessary PlatformIO system data (temporary files, unnecessary packages, etc.).
Use `pio system prune --dry-run` to list them or `pio system prune` to save disk space.
INFO Successfully compiled program.
INFO Resolving IP address of deye-c.solar.MYDOMAIN.TLD
INFO Connecting to 1xx.xx.xx.3 port 3232...
INFO Connected to 1xx.xx.xx.3
INFO Uploading /home/USER/ESPHome/esphome-for-deye/.esphome/build/deye-c/.pioenvs/deye-c/firmware.bin (1058592 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 5.31 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from deye-c.solar.MYDOMAIN.TLD using esphome API
INFO Successfully connected to deye-c @ 1xx.xx.xx.3 in 5.138s
INFO Successful handshake with deye-c @ 1xx.xx.xx.3 in 0.123s
[18:39:49][I][app:115]: ESPHome version 2025.5.1 compiled on May 28 2025, 18:38:46
[18:39:49][I][app:117]: Project luckylinux.esphome-for-deye version 20250528
```

And from Home Assistant after the updated Firmware was uploaded:
```
Firmware: 20250528 (ESPHome 2025.5.1)
Hardware: 20240820 
```

> [platform] is an invalid option for [ota]. Please check the indentation. platform: esphome
> 
> [version] is an invalid option for [ota]. Please check the indentation. version: 2 password: !secret ota_password safe_mode: [source config/safe_mode.yaml:1]
> 
> Component safe_mode cannot be loaded via YAML (no CONFIG_SCHEMA). {} async_tcp: None
> 
> This feature is only available with frameworks ['arduino']. {} In case of errors, it is suggested to run: esphome clean.
> 
> Have any more ideas?

Just use some standard full Fledged Linux Distributionn ?

I already had to deal with some Users having weird Issues on e.g. Truenas while trying to run some compiled Python Code using `nuitka`. Anecdotal but I am sure Synology is adding their own Issues on top of whatever *might* exist in my Code. Whether it's some Apparmor/SELinux Issue due to misconfiguration Permission, some weird Mount Options, running in LXC Container, etc.

These Appliances might be nice, fun and easy for the Users, but they add their own set of Issues which, since you don't even know how such Systems works to begin with, is even harder to troubleshoot.

So as I said, either:
a. Take some Virtualization Setup that should be available on MacOS X (in no particular Order: UTM Virtualization Solution, Parallels Desktop, Virtualbox, ...), create a Thin Provisioned (so that it does NOT occupy the full size unless needed) Virtual Disk of say 32GB, install e.g. Ubuntu Linux on it, then pass-through the USB Device to it (the M5 Stack Programmer Chip included within the M5 Stack ESP32-S3 Device)
b. Take a USB Flash Drive of 16GB or more (preferably 32GB or more), install e.g. Ubuntu Linux on it, then reboot your Mac and hopefully you can boot from LiveUSB if the BIOS/UEFI on that Locked Down System allows you to ...

Because I sure cannot provide help blindly without any clear logs.

You copy files here and there without using my Script ? That's fine but then it's up to you to make sure the folder Structure is consistent and the Variables replaced.

What do you think this `for` Loop does ?
https://github.com/luckylinux/esphome-for-deye/blob/main/build.sh#L82

```
files=$(find ./ -iname "*.yaml")
for file in $files
do
   replace_text ./$file "board" "${board}"
   replace_text ./$file "variant" "${variant}"
   replace_text ./$file "mqtt_topic_prefix" "${topic_prefix}"
done
```

It replaces some Variables in the "Templates" with their Values.
The Replaced Variables include:
- `board`
- `variant`
- `mqtt_topic_prefix`

Remember that also `secrets.sh` generates the `secrets.yaml` File that contains a lot of Variables (`ota_password` for one, which might explain your first Error).

tmbreuker, 2025-05-28:
@luckylinux so the mqtt host is not running on the atom? My bad then, I'll comment it out. 

I removed `esphome::` **not** `::esphome` from lines 419 and 425. You might be right about the hardware dependency.

tmbreuker, 2025-05-28:
sure, I just check my inverter. There's already another system connected, is it possible to just add the wires for the atom?

[image]

luckylinux, 2025-05-28:
What would that solve ? The problem is that you have another Master I believe. You can try and should check, but IMHO it would NOT make sense that are two Slaves (but you never know ...).

I don't know what those Wires (that are already occupying the MODBUS Port) are connected to.

Is it the EASTRON (or similar) Meter that has been wrongly been connected to the MODBUS Port instead of the METER Port ?

According to my Manual the METER 485 Connector should be the 3rd RJ45 Jack from the Left. Yours is the 4th one (MODBUS).

But then again you should know (or the one that did the Installation) where those Wires are coming from / going to.

luckylinux, 2025-05-28:
Maybe start reading what I write ...

tmbreuker, 2025-05-28:
> I don't know what those Wires (that are already occupying the MODBUS Port) are connected to. 
I do, this is connected to a commercial EMS (bliq) which I currently deactivated.

> 
> Is it the EASTRON (or similar) Meter that has been wrongly been connected to the MODBUS Port instead of the METER Port ?
> 
> According to my Manual the METER 485 Connector should be the 3rd RJ45 Jack from the Left. Yours is the 4th one (MODBUS).

Remember, I have a 10K and you have a 12K, that might be slightly different

tmbreuker, 2025-05-29:
So, I removed bliq and connected ESP32 and it is working :). Now I need to see how can i add the dashboard as I am also a HA newby

luckylinux, 2025-05-29:
As expected 😉. Good News nevertheless

tmbreuker, 2025-05-29:
@luckylinux that is good news. In the introduction of read.me there's a wonderful dashboard which I would love to replicate. I am rather clueless. Is it possible to set this up through a copy-paste (incl adjustements, renaming entity names) of a yaml which you can share or do I need to set this up manually?

tmbreuker, 2025-05-29:
wow, this is an amazing guide. I just stumble upon an oddity in the multipl_entity_row: It doesn't show me From, To and SOC:

[image]

luckylinux, 2025-05-29:
Probably you can set the Battery to `No Battery` instead of Lithium but that will cause a Restart of the Inverter to reconfigure.

Easier probably to set Discharge Current to Zero instead.

luckylinux, 2025-05-29:
Did you at least Check that you have the Correct Settings on the Inverter ?

[image]
I believe you do NOT want to keep `Grid Signal` Ticked as in the Picture, I would imagine that requires some Signal to tell it to Charge.

Try maybe 95% instead of 100% for SOC.

tmbreuker, 2025-05-29:
> Probably you can set the Battery to `No Battery` instead of Lithium but that will cause a Restart of the Inverter to reconfigure.
> 
> Easier probably to set Discharge Current to Zero instead.

Yeah, I was checking the documentation and thinking of: using register
109  Maximum battery discharge current Max A discharge R/W [0,185] 1A 0-185A

I currently only use PV to charge.

tmbreuker, 2025-05-29:
I noticed this is already available : sun12k_battery_maximum_discharge_current - so setting up automation was a piece of cake for me.

tmbreuker, 2025-05-29:
> You are not providing any Values here so I think I'll pass.
> 
What do you mean by this?

> There is a register to set the Zero-Power to use for the CTs for instance. If it's a small Discharge, maybe it's that Parameter.
> 
I don't particularly fancy the interface of deye as all these acronyms mean so little to me, like CT. Don't know what this means without google-ing this first.

This image you shared also provides another option: Ticking "Time of use" on and off could also do the trick, provided Grid is there, which in your case wouldn't help.

[https://private-user-images.githubusercontent.com/7126291/448966384-c1cf97c5-412a-4d60-bd46-41e2a808a1ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg1NDgwOTgsIm5iZiI6MTc0ODU0Nzc5OCwicGF0aCI6Ii83MTI2MjkxLzQ0ODk2NjM4NC1jMWNmOTdjNS00MTJhLTRkNjAtYmQ0Ni00MWUyYTgwOGExZWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MjlUMTk0MzE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjFlNDk0NmQ3MTBhNWMyZjgwZDY3ZTIzMmI3YWEyMWFmMTIwYWI2OTAxNDFlMWI3NTRmNmE0YmQxMDQ2ZjVmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.rFB8UgChqiQjf4c4lnaeN1AevuhCm_XkF0Wx63lf9ME](url)

tmbreuker, 2025-05-31:
@luckylinux  could I use your build https://github.com/luckylinux/esphome-for-deye/blob/main/build.sh to setup ESP32 as a slave instead of a master, knowing that I also need to change my Inverter to master mode

luckylinux, 2025-05-31:
@tmbreuker: well you could but you'll need to pretty much change EVERYTHING in the YAML Files.

If this is to get your EMS working at the same Time, I'm not even sure if it's worth the Trouble.

As I said, you could pass the Information from the ESP32-S3 to the Bliq EMS either via MQTT and a second ESP32-S3 and another RS485 Transceiver (or possibly from thew same ESP32-S3, although you might get short on Memory/Flash and run into Stability Issues) and do like this:
MQTT/HA <--> ESP32-S3 <--> (Master) RS485 Transceiver <--> Deye Inverter (Slave)
                      <--> (Slave) RS485 Transceiver  <--> Bliq EMS (Master)

OR
MQTT/HA <--> ESP32-S3 <--> (Master) RS485 Transceiver <--> Deye Inverter (Slave)
MQTT/HA <--> ESP32-S3 <--> (Slave) RS485 Transceiver  <--> Bliq EMS (Master)

Not sure how the Deye would run in Master Mode, assuming it's even possible. And not sure exactly how you would be able to set Settings there. Granted the Deye could just do a "Pull" of the Settings instead of the ESP32-S3 doing a "Push", but I'm not sure if that is supported, reliable. There could be many things that start going wrong if it's supported (probably because the Deye will try to pull ALL Registers/Settings, so what happens for those that are not implemented or not implemented correctly ?).

I do NOT recommend to do that but ... your Choice 😃.

tmbreuker, 2025-05-31:
@luckylinux thanks, it is running now, without Bliq EMS and as you said 'don't change a running system'. I need to think this through and decide what is the better option for me as I am not familiar with MQTT.

luckylinux, 2025-05-31:
@tmbreuker You can probably do everything on a single ESP32-S3 (or between two ESP32-S3) without MQTT/HA, but it's probably easier to do with MQTT/HA actually for the potential Issue due to Resources (Flash/RAM) and Load.

Think well before you decide to embark on that Journey ! On one hand, you know how to emulate a Deye Inverter (the Registers are described in the `reference` Folder and implemented in the [Deye 3p modbus address list.docx](https://github.com/klatremis/esphome-for-deye/blob/main/Deye%203p%20modbus%20address%20list.docx)), so to implement a Slave ESP32-S3 to talk to the Bliq EMS shouldn't be that complicated.

On the other Hand for a beginner (or even me, I never used the `modbus_controller` as a Slave), it's going to take some Time. But once you figure out how to make it work for 1 Register or 2, then it should be easy to scale it up to the entire range of Registers.

https://esphome.io/components/modbus

You will probably change from `role: client` (default if not Specified) to `role: server` if you want to do that.

https://esphome.io/components/modbus_controller.html

Note that running the ESP32-S3 as a Client means that the ESP32-S3 is the Master, while running the ESP32-S3 as a Server means that the ESP32-S3 is the Slave !!!

>> The modbus_controller component creates a RS485 connection to either:
>> - control a Modbus server (slave) device, letting your ESPHome node to act as a Modbus client (master). You can access the coils, inputs, holding, read registers from your devices as sensors, switches, selects, numbers or various other ESPHome components and present them to your favorite Home Automation system. You can even write them as binary or float ouptputs from ESPHome.
>> - let your ESPHome node act as a Modbus server, allowing a ModBUS client to read data (like sensor values) from your ESPHome node.
>> 
>> To choose the role, set the role attribute of the [Modbus Component](https://esphome.io/components/modbus) upon which this modbus_controller component relies. client is the default.

tmbreuker, 2025-05-31:
@luckylinux My gut feelig tells me this is the best solution:
MQTT/HA <--> ESP32-S3 <--> (Master) RS485 Transceiver <--> Deye Inverter (Slave)
MQTT/HA <--> ESP32-S3 <--> (Slave) RS485 Transceiver <--> Bliq EMS (Master)

I envision these steps:
1. Install MQTT on HA: I used the Mosquito broker from HA app store
2. Setup S3/RS485 master to send/receive to HA MQTT: this time I do need to build with mqtt - i know how to do so now :)
3. Order another Atom S3 and RS485 transceiver: been there, done that
4. Setup S3 and RS485 as slave and send/receive to HA MQTT: this is the tricky part, but I'll give it a go

luckylinux, 2025-05-31:
@tmbreuker Yeah if Reliability is the Target (IMHO **Secondary** Monitoring & Datalogging < Primary Control & Measurement) and for Modularity it's easier to use a second Independent Device.

It shouldn't be that complicated, but the Devil is in the Details. You basically have to do everything in Reverse:
- Modbus Server (SLAVE) instead of Modbus Client (MASTER)
- Read from MQTT/HA instead of Writing to MQTT/HA
- All Control Stuff (write Setting to Deye) will need to be removed or made Read-Only as the Bliq EMS is only reading

# Modbus Server - Decide what to expose
What you want to expose to the BLIQ EMS will have to be done using a Modbus Server similar to the Example here
https://esphome.io/components/modbus_controller#example-server

In particular (and a bit expanded) determine which Registers you want to expose to the BLIQ EMS
```
modbus_controller:
  - modbus_id: modbus_server
    address: 0x4
    server_registers:
      - address: 0x0002
        value_type: S_DWORD_R
        read_lambda: |-
          return id(evse_voltage_l1).state;
      - address: 0x0004
        ...
        ...
      - ...
```

That's the first Part.

# Getting the Data from MQTT/HomeAssistant

The Second Part is of course to read via HA/MQTT what the other ESP32 Sends, that seems even more straightforward to be honest.

## Using MQTT

https://esphome.io/components/sensor/mqtt_subscribe.html

e.g.
```
sensor:
  - platform: mqtt_subscribe
    name: "Data from topic"
    id: mysensor
    topic: the/topic
```

(you can use a Docker/Podman Container running `mqtt-explorer` in order to Visually see what's going on, which Entity to take, etc, but of course you must run your own `mqtt` Broker/Server first, e.g. Eclipse Mosquitto MQTT, but basically the Topic is the ENTIRE Path up to and including the Entity ID / Sensor Name).

## Using Homeassistant API

Whereas if you only use HomeAssistant API without MQTT
https://esphome.io/components/sensor/homeassistant.html

(also some useful Notes here: https://community.home-assistant.io/t/fetch-data-from-api-and-present-it-on-a-display-fails/359663)

For Instance
```
sensor:
  - platform: homeassistant
    id: current_temperature
    entity_id: climate.living_room
    attribute: current_temperature
```

# General Remarks
Instead of only using the `switch`, maybe ADD `binary_switch` which is Read only:
https://esphome.io/components/binary_sensor/switch.html

Note that I added and on the Switch itself which **SHOULD** prevent it from being visible as something that you can change.

https://esphome.io/components/sensor/homeassistant.html#configuration-variables

```
binary_sensor:
  - platform: my_sensor
    name: "My Name"
    source_id: my_sensor_internal

switch:
  - platform: homeassistant
    disabled_by_default: true
    internal: true
    id: my_sensor_internal
    entity_id: FULL_Entity_ID_Including_OTHER_ESP32-S3_Prefix_Plus_Full_Entity_Path
    # attribute: state # (or don't set, according to Documentation it should default to state)

```

Maybe other Sensors need to be modified as well, but as a general Rule I'd expect:
- `sensor` -> `sensor` (it's read only already)
- `number` -> `sensor` (you don't need to change Settings from that ESP32-S3)
- `switch` -> `binary_sensor` (you don't need to change Settings from that ESP32-S3)

Rinse and repeat approx. 200 Times (for both the `modbus_server` and the `MQTT` or `homeassistant` Parts) 😄.