klatremis/esphome-for-deye #50: ESP Builder 2025.7 and higher

Redosch, 2025-09-21:
Hello everyone,

I have been reading and writing values via MODBUS from my Deye Sun-12K for over 15 months, as described in the example, using an ESP32.
Since the 2025.7 update, I am no longer receiving any values. The same problem also occurs with versions 2025.8 and 2025.9.
Does anyone have any ideas?

rvdgaag, 2025-10-05:
I have similar issue, from one moment to the other it stopped working. I think it has to do with HA updates in the core. Restoring  the ESPBuilder to older versions did not help.

For this moment I have setup the Solarman integration which works slow but fine, prefer to go back to the klatremis solution with esp32.

Redosch, 2025-10-05:
Hello rvdgaag,

In my HAS system, downgrading to ESP Builder 2025.7 worked until mid-September 2025. But now downgrading do not fixes the problem. 

Yes, the HACS integration Solarman works well, but I am missing a few data points. Do you know how to add more data points in Solarman?

rvdgaag, 2025-10-05:
> Yes, the HACS integration Solarman works well, but I am missing a few data points. Do you know how to add more data points in Solarman?

No I don't but you can always do a request on Github